espressif/vscode-esp-idf-extension #1550: Can't compile

flair0626, 2025-05-13:
### OS

Windows

### Operating System version

Win10

### Visual Studio Code version

1.100.0

### ESP-IDF version

v5.4.1

### Python version

3.11.2

### Doctor command output

[report.txt](https://github.com/user-attachments/files/20180109/report.txt)

### Description

Unable to compile with the error message "Something went wrong while trying to build the project".

[image]

### Debug Message

None.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2025-05-13:
Based on the doctor command, I can see that CMake and Ninja are not found by the extension process because if you run the command `where cmake` it will fail.
Also I can see that powershell can't be spawn for some reason.

I suspect there is something in your environment variable PATH that is not giving access to cmake or ninja (and powershell)

Run the `ESP-IDF: Open ESP-IDF Terminal` and type `where cmake` and see what is the result.

flair0626, 2025-05-13:
使用 idf.py build 命令可以正常编译

brianignacio5, 2025-05-15:
This tells me nothing I can work with.

> Run the ESP-IDF: Open ESP-IDF Terminal and type where cmake and see what is the result.

What is the result of this ?

Can you reproduce the issue in other computer ? Could you provide me with some steps to reproduce the issue as I can't reproduce the issue on Windows 11 without any additional information.

flair0626, 2025-05-15:
[image]

brianignacio5, 2025-05-15:
If you set the full path of course is going to work. 

Try just `cmake.exe`

flair0626, 2025-05-15:
[image]

brianignacio5, 2025-05-21:
I'm sorry @flair0626 but I can't reproduce in our Windows test machines.

There is something in your environment setup that is preventing the vscode extension to find cmake and ninja in PATH. If you could provide an example project or provide me steps to reproduce the issue in a different machine it would help me to find the issue and provide a solution.

flair0626, 2025-05-21:
This is an example project.
[project-name.zip](https://github.com/user-attachments/files/20358679/project-name.zip)
These are all the environment variables.
[environment.txt](https://github.com/user-attachments/files/20358678/environment.txt)

closed 2025-06-04 by brianignacio5