espressif/vscode-esp-idf-extension #1561: [Feature Request]: open disassembly view at specified C source line

bryghtlabs-richard, 2025-05-20:
**Is your feature request related to a problem? Please describe.**
I'd like to open the disassembly view for a specific function that is not in the call stack.

**Describe the solution you'd like**
I'd like right-click -> open disassembly view on a function name to take me to that function, or right-click -> open disassembly view on a non-function-call line to take me to that function's disassembly view

**Describe alternatives you've considered**
Sometimes I set a breakpoint on the code I want to disassemble then run until I can hit it, but this changes the program state. Sometimes I can scroll in the disassembly view. But usually I use objdump instead.

**Additional context**
The documentation states, `You can check the assembly code during a debugging session by right-clicking on any line in in a source code file and selecting Open Disassembly View. Disassemble View shows the assembly code with C code, where you can also set breakpoints.` but this always seems to open the disassembly view at either the current program counter, or the previous location the disassembly view was opened at.

Please consider this a low-priority request.

brianignacio5, 2025-05-27:
Consider that the Open Disassembly view command will send the current program counter in the debug adapter disassemble request. https://microsoft.github.io/debug-adapter-protocol/specification#Requests_Disassemble

For this feature, vscode should add the right click on a function name to trigger the disassembly on that memory reference.

I think this feature is related more to objdump or memory view instead, since disassembly view is using gdb and probably referring to the current program execution. I think a memory view feature is more what you would like to see here and that is a separated feature to implement all together.

bryghtlabs-richard, 2025-05-27:
>I think a memory view feature is more what you would like to see here and that is a separated feature to implement all together.

It could work that way. It would need to be aware of the 2B/3B xtensa instructions though, since you need to start from the function, then disassemble according to each instruction length.

From disassembly view, if I ctrl-click into a function call and setup a breakpoint in that function, then hit run, I wouldn't like to have to change windows to see what happened. I think a debug/disassembly-view is probably best here.

closed 2025-06-30 by github-actions[bot]

bryghtlabs-richard, 2025-06-30:
Note: this was not completed, but was transferred to VSCode repository, as it's mostly dependent on VSCode passing the disassembly target address into the disassembly viewer(and without this, ESP-VSCode couldn't do anything).