espressif/vscode-esp-idf-extension #1562: [Feature Request]: ability to ctrl-click in disassembly view

bryghtlabs-richard, 2025-05-20:
**Is your feature request related to a problem? Please describe.**
It seems like opening disassembly-view is a one-way ticket. You can't really use it to navigate functions(other than scrolling to nearby functions).

**Describe the solution you'd like**
I'd like the ability to `ctrl-click` here, and have it navigate the disassembly view like it does the regular source view.

[image]

**Describe alternatives you've considered**
You can switch back to high-level source view, but see https://github.com/espressif/vscode-esp-idf-extension/issues/1561 .

**Additional context**
Please consider this a low-priority request.

brianignacio5, 2025-06-04:
I think this is a vscode itself issue, not specific to this extension.

bryghtlabs-richard, 2025-06-04:
I think it may be also. If vscode isn't sending the request for the selected line when I ctrl-click, it'll need to before you could do anything with that information.