espressif/vscode-esp-idf-extension #1569: VS code (reading 'fsPath')

Bazconnected, 2025-05-23:
### OS

MacOS

### Operating System version

macOs 15.5

### Visual Studio Code version

1.100

### ESP-IDF version

5.4.1

### Python version

3.13

### Doctor command output

when try to creat new project get this error 
Cannot read properties of undefined (reading 'fsPath')

### Description

when try to creat new project get this error 
Cannot read properties of undefined (reading 'fsPath')

### Debug Message

when try to creat new project get this error 
Cannot read properties of undefined (reading 'fsPath')

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

closed 2025-05-27 by brianignacio5

MatheusCaltran, 2025-06-08:
I saw some comments that Reverting to extension 1.9.x solves the problem

brianignacio5, 2025-06-09:
Or using the GitHub master vsix from [here](https://nightly.link/espressif/vscode-esp-idf-extension/workflows/ci/master/esp-idf-extension.vsix.zip) until the next release @MatheusCaltran since #1538 is already merged in master.