espressif/vscode-esp-idf-extension #1571: Sdkconfig not found when trying to flash device

renzenicolai, 2025-05-24:
### OS

Linux

### Operating System version

Linux

### ESP-IDF version

5.5.0

### Python version

3.13.2

### Doctor command output

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS linux x64 6.13.8-arch1-1 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 <HOMEPATH>/.nix-profile/bin:/nix/var/nix/profiles/default/bin:/usr/local/sbin:/usr/local/bin:/usr/bin:/usr/lib/jvm/default/bin:/usr/bin/site_perl:/usr/bin/vendor_perl:/usr/bin/core_perl:/opt/rocm/bin:/usr/lib/rustup/bin 
System environment variable PYTHON 
 undefined 
Visual Studio Code Remote name undefined 
Visual Studio Code version 1.99.2 
Visual Studio Code language en 
Visual Studio Code shell /usr/bin/zsh 
ESP-IDF Extension version 1.10.0 
Workspace folder <HOMEPATH>/Documents/Nicolai_Electronics/Tanmatsu/Apps/legacy_radio_fw 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) <HOMEPATH>/Documents/Nicolai_Electronics/Tanmatsu/Apps/legacy_radio_fw/esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
ESP-HomeKit-SDK Path (idf.espHomeKitSdkPath) ${env:HOMEKIT_PATH}
Custom extra paths <HOMEPATH>/Documents/Nicolai_Electronics/Tanmatsu/Apps/legacy_radio_fw/esp-idf-tools/tools/xtensa-esp-elf-gdb/16.2_20250324/xtensa-esp-elf-gdb/bin:<HOMEPATH>/Documents/Nicolai_Electronics/Tanmatsu/Apps/legacy_radio_fw/esp-idf-tools/tools/riscv32-esp-elf-gdb/16.2_20250324/riscv32-esp-elf-gdb/bin:<HOMEPATH>/Documents/Nicolai_Electronics/Tanmatsu/Apps/legacy_radio_fw/esp-idf-tools/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin:<HOMEPATH>/Documents/Nicolai_Electronics/Tanmatsu/Apps/legacy_radio_fw/esp-idf-tools/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin:<HOMEPATH>/Documents/Nicolai_Electronics/Tanmatsu/Apps/legacy_radio_fw/esp-idf-tools/tools/esp32ulp-elf/2.38_20240113/esp32ulp-elf/bin:<HOMEPATH>/Documents/Nicolai_Electronics/Tanmatsu/Apps/legacy_radio_fw/esp-idf-tools/tools/cmake/3.30.2/bin:<HOMEPATH>/Documents/Nicolai_Electronics/Tanmatsu/Apps/legacy_radio_fw/esp-idf-tools/tools/openocd-esp32/v0.12.0-esp32-20250422/openocd-esp32/bin:<HOMEPATH>/Documents/Nicolai_Electronics/Tanmatsu/Apps/legacy_radio_fw/esp-idf-tools/tools/ninja/1.12.1:<HOMEPATH>/Documents/Nicolai_Electronics/Tanmatsu/Apps/legacy_radio_fw/esp-idf-tools/tools/esp-rom-elfs/20241011
ESP-IDF extra vars
    OPENOCD_SCRIPTS: <HOMEPATH>/Documents/Nicolai_Electronics/Tanmatsu/Apps/legacy_radio_fw/esp-idf-tools/tools/openocd-esp32/v0.12.0-esp32-20250422/openocd-esp32/share/openocd/scripts
    ESP_ROM_ELF_DIR: <HOMEPATH>/Documents/Nicolai_Electronics/Tanmatsu/Apps/legacy_radio_fw/esp-idf-tools/tools/esp-rom-elfs/20241011/
    IDF_PYTHON_ENV_PATH: <HOMEPATH>/Documents/Nicolai_Electronics/Tanmatsu/Apps/legacy_radio_fw/esp-idf-tools/python_env/idf5.5_py3.13_env
    ESP_IDF_VERSION: 5.5
    PATH: <HOMEPATH>/Documents/Nicolai_Electronics/Tanmatsu/Apps/legacy_radio_fw/esp-idf-tools/tools/xtensa-esp-elf-gdb/16.2_20250324/xtensa-esp-elf-gdb/bin:<HOMEPATH>/Documents/Nicolai_Electronics/Tanmatsu/Apps/legacy_radio_fw/esp-idf-tools/tools/riscv32-esp-elf-gdb/16.2_20250324/riscv32-esp-elf-gdb/bin:<HOMEPATH>/Documents/Nicolai_Electronics/Tanmatsu/Apps/legacy_radio_fw/esp-idf-tools/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin:<HOMEPATH>/Documents/Nicolai_Electronics/Tanmatsu/Apps/legacy_radio_fw/esp-idf-tools/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin:<HOMEPATH>/Documents/Nicolai_Electronics/Tanmatsu/Apps/legacy_radio_fw/esp-idf-tools/tools/esp32ulp-elf/2.38_20240113/esp32ulp-elf/bin:<HOMEPATH>/Documents/Nicolai_Electronics/Tanmatsu/Apps/legacy_radio_fw/esp-idf-tools/tools/openocd-esp32/v0.12.0-esp32-20250422/openocd-esp32/bin:<HOMEPATH>/Documents/Nicolai_Electronics/Tanmatsu/Apps/legacy_radio_fw/esp-idf-tools/python_env/idf5.5_py3.13_env/bin:<HOMEPATH>/Documents/Nicolai_Electronics/Tanmatsu/Apps/legacy_radio_fw/esp-idf/tools:$PATH
    IDF_DEACTIVATE_FILE_PATH: /tmp/tmp_0apvg8nidf_45237
User extra vars (idf.customExtraVars)
    IDF_TARGET: esp32c6
System python Path (idf.pythonInstallPath) /usr/bin/python
Virtual environment Python path (computed) <HOMEPATH>/Documents/Nicolai_Electronics/Tanmatsu/Apps/legacy_radio_fw/esp-idf-tools/python_env/idf5.5_py3.13_env/bin/python
Serial port (idf.port) /dev/ttyACM1
OpenOCD Configs (idf.openOcdConfigs) board/esp32c6-builtin.cfg
ESP-IDF Tools Path (idf.toolsPath) <HOMEPATH>/Documents/Nicolai_Electronics/Tanmatsu/Apps/legacy_radio_fw/esp-idf-tools
Git Path (idf.gitPath) /usr/bin/git
Notification Mode (idf.notificationMode) All
Flash type (idf.flashType) 
Flash partition to use (idf.flashPartitionToUse) all
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-HomeKit Path (idf.espHomeKitSdkPath) false
Access to ESP-IDF Custom extra paths
Access to <HOMEPATH>/Documents/Nicolai_Electronics/Tanmatsu/Apps/legacy_radio_fw/esp-idf-tools/tools/xtensa-esp-elf-gdb/16.2_20250324/xtensa-esp-elf-gdb/bin: true
Access to <HOMEPATH>/Documents/Nicolai_Electronics/Tanmatsu/Apps/legacy_radio_fw/esp-idf-tools/tools/riscv32-esp-elf-gdb/16.2_20250324/riscv32-esp-elf-gdb/bin: true
Access to <HOMEPATH>/Documents/Nicolai_Electronics/Tanmatsu/Apps/legacy_radio_fw/esp-idf-tools/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin: true
Access to <HOMEPATH>/Documents/Nicolai_Electronics/Tanmatsu/Apps/legacy_radio_fw/esp-idf-tools/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin: true
Access to <HOMEPATH>/Documents/Nicolai_Electronics/Tanmatsu/Apps/legacy_radio_fw/esp-idf-tools/tools/esp32ulp-elf/2.38_20240113/esp32ulp-elf/bin: true
Access to <HOMEPATH>/Documents/Nicolai_Electronics/Tanmatsu/Apps/legacy_radio_fw/esp-idf-tools/tools/cmake/3.30.2/bin: false
[... 156 lines cut ...]
  "code": 1,
  "killed": false,
  "signal": null,
  "cmd": "<HOMEPATH>/Documents/Nicolai_Electronics/Tanmatsu/Apps/legacy_radio_fw/esp-idf-tools/python_env/idf5.5_py3.13_env/bin/python <HOMEPATH>/Documents/Nicolai_Electronics/Tanmatsu/Apps/legacy_radio_fw/esp-idf/tools/check_python_dependencies.py -r <HOMEPATH>/.vscode/extensions/espressif.esp-idf-extension-1.10.0/esp_debug_adapter/requirements.txt"
}
--------------------------------------------------------------------------------------------------------------------------------------------
----------------------------------------------------------- Logfile -----------------------------------------------------------------
{"message":"Cannot access filePath: \"sdkconfig_tanmatsu\"","stack":"Error: ENOENT: no such file or directory, access '\"sdkconfig_tanmatsu\"'\n    at Object.accessSync (node:fs:245:11)\n    at Object.t.accessSync (node:electron/js2c/node_init:2:9332)\n    at at (/home/renze/.vscode/extensions/espressif.esp-idf-extension-1.10.0/dist/extension.js:2:1453125)\n    at yt (/home/renze/.vscode/extensions/espressif.esp-idf-extension-1.10.0/dist/extension.js:2:1455487)\n    at Ze (/home/renze/.vscode/extensions/espressif.esp-idf-extension-1.10.0/dist/extension.js:2:1450879)\n    at zt.configureWithDefaultValues (/home/renze/.vscode/extensions/espressif.esp-idf-extension-1.10.0/dist/extension.js:2:1469429)","category":"src utils canAccessFile","level":"error","timestamp":"2025-05-24T09:10:35.588Z"}
...

### Description

It appears the plugin is trying to put extra quotes around the filename of the sdkconfig file

### Debug Message

{"message":"Cannot access filePath: \"sdkconfig_tanmatsu\"","stack":"Error: ENOENT: no such file or directory, access '\"sdkconfig_tanmatsu\"'\n    at Object.accessSync (node:fs:245:11)\n    at Object.t.accessSync (node:electron/js2c/node_init:2:9332)\n    at at (/home/renze/.vscode/extensions/espressif.esp-idf-extension-1.10.0/dist/extension.js:2:1453125)\n    at yt (/home/renze/.vscode/extensions/espressif.esp-idf-extension-1.10.0/dist/extension.js:2:1455487)\n    at Ya (/home/renze/.vscode/extensions/espressif.esp-idf-extension-1.10.0/dist/extension.js:2:169664 [... cut ...] vw.h (file:///usr/lib/code/out/vs/workbench/api/node/extensionHostProcess.js:111:41495)","category":"src utils canAccessFile","level":"error","timestamp":"2025-05-24T09:33:23.183Z"}

### Other Steps to Reproduce

I want to use a custom sdkconfig path so I entered an sdkconfig path in the project configuration menu

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

renzenicolai, 2025-05-24:
Interesting bit of extra info:

Turns out the extension uses a function that parses CMakeLists.txt to determine the sdkconfig file location. I had this in there:

`set(SDKCONFIG "sdkconfig_tanmatsu")`

Removing the quotes does remove the unwanted escaped quotes you see in the logfile above.

`set(SDKCONFIG sdkconfig_tanmatsu)`

But the extension still can not find the file even though it does exist in the main workspace folder, next to the CMakeLists.txt file which it clearly does find.

---

Other question in the same direction: why doesn't the extension just use the sdkconfig location as provided in the project configuration?

brianignacio5, 2025-05-27:
The issue is that you just set the name not the path.

Please look at https://github.com/espressif/esp-idf/blob/master/examples/build_system/cmake/multi_config/CMakeLists.txt#L6

`set(SDKCONFIG "${CMAKE_BINARY_DIR}/sdkconfig_tanmatsu")` instead of `set(SDKCONFIG "sdkconfig_tanmatsu")`

You can also set in Project configuration the sdkconfig path and not directly in CMakeLists.txt set(SDKCONFIG "sdkconfig_tanmatsu") by using the sdkconfig field and set the full sdkconfig path.

brianignacio5, 2025-06-09:
Closing this issue due to lack of response. Comment here if you need more assistance.