espressif/vscode-esp-idf-extension #1572: Cant' get JTAG interface running on several different devices

NoPlayBack, 2025-05-25:
### OS

Windows

### Operating System version

Win 11

### Visual Studio Code version

1.100.2

### ESP-IDF version

1.10.0

### Python version

3

### Doctor command output

{"message":"OpenOCD server failed to start because of Error: libusb_open() failed with LIBUSB_ERROR_NOT_FOUND\n❌ Error: libusb_open() failed with LIBUSB_ERROR_NOT_FOUND\r\n","stack":"Error: OpenOCD server failed to start because of Error: libusb_open() failed with LIBUSB_ERROR_NOT_FOUND\n\tat Socket.<anonymous> (\\dist\\extension.js:2:1576120)\n\tat Socket.emit (node:events:524:28)\n\tat addChunk (node:internal/streams/readable:561:12)\n\tat readableAddChunkPushByteMode (node:internal/streams/readable:512:3)\n\tat Readable.push (node:internal/streams/readable:392:5)\n\tat Pipe.onStreamRead (node:internal/stream_base_commons:191:23)","category":"OpenOCDManager stderr","level":"error","timestamp":"2025-05-25T12:15:29.962Z"}
{"message":"OpenOCD server failed to start because of Error: unable to open ftdi device with description '*', serial '*' at bus location '*'\n❌ Error: unable to open ftdi device with description '*', serial '*' at bus location '*'\r\nC:\\Users\\Eddie\\.espressif\\tools\\openocd-esp32\\v0.12.0-esp32-20241016\\openocd-esp32\\share\\openocd\\scripts/target/esp_common.cfg:9: Error: \r\nat file \"C:\\Users\\Eddie\\.espressif\\tools\\openocd-esp32\\v0.12.0-esp32-20241016\\openocd-esp32\\share\\openocd\\scripts/target/esp_common.cfg\", line 9\r\n","stack":"Error: OpenOCD server failed to start because of Error: unable to open ftdi device with description '*', serial '*' at bus location '*'\n\tat Socket.<anonymous> (\\dist\\extension.js:2:1576120)\n\tat Socket.emit (node:events:524:28)\n\tat addChunk (node:internal/streams/readable:561:12)\n\tat readableAddChunkPushByteMode (node:internal/streams/readable:512:3)\n\tat Readable.push (node:internal/streams/readable:392:5)\n\tat Pipe.onStreamRead (node:internal/stream_base_commons:191:23)","category":"OpenOCDManager stderr","level":"error","timestamp":"2025-05-25T12:15:29.963Z"}
{"message":"OpenOCD Exit with non-zero error code 1","stack":"Error: Spawn exit with non-zero1\n\tat ChildProcess.<anonymous> (\\dist\\extension.js:2:1576813)\n\tat ChildProcess.emit (node:events:524:28)\n\tat maybeClose (node:internal/child_process:1104:16)\n\tat ChildProcess._handle.onexit (node:internal/child_process:304:5)","category":"OpenOCDManager close","level":"error","timestamp":"2025-05-25T12:15:29.968Z"}
{"user":true,"level":"warn","message":"Can't perform JTag flash, because OpenOCD server is not running!","timestamp":"2025-05-25T12:15:30.895Z"}
{"command":"C:\\Users\\Eddie\\.espressif\\python_env\\idf5.4_py3.11_env\\Scripts\\python.exe","message":"non zero exit code null\r\n\r\nesptool.py v4.8.1\r\nSerial port COM1\r\nConnecting.........","stack":"Error: non zero exit code null\r\n\r\nesptool.py v4.8.1\r\nSerial port COM1\r\nConnecting.........\n\tat ChildProcess.<anonymous> (\\dist\\extension.js:2:1452925)\n\tat ChildProcess.emit (node:events:524:28)\n\tat ChildProcess._handle.onexit (node:internal/child_process:293:12)","category":"src utils spawn","level":"error","timestamp":"2025-05-25T12:16:44.706Z"}
{"command":"C:\\Users\\Eddie\\.espressif\\python_env\\idf5.4_py3.11_env\\Scripts\\python.exe","message":"non zero exit code null\r\n\r\nesptool.py v4.8.1\r\nSerial port COM12\r\nConnecting.........","stack":"Error: non zero exit code null\r\n\r\nesptool.py v4.8.1\r\nSerial port COM12\r\nConnecting.........\n\tat ChildProcess.<anonymous> (\\dist\\extension.js:2:1452925)\n\tat ChildProcess.emit (node:events:524:28)\n\tat ChildProcess._handle.onexit (node:internal/child_process:293:12)","category":"src utils spawn","level":"error","timestamp":"2025-05-25T12:16:44.707Z"}
{"message":"OpenOCD server failed to start because of Error: JTAG scan chain interrogati\n❌ Error: JTAG scan chain interrogati","stack":"Error: OpenOCD server failed to start because of Error: JTAG scan chain interrogati\n\tat Socket.<anonymous> (\\dist\\extension.js:2:1576120)\n\tat Socket.emit (node:events:524:28)\n\tat addChunk (node:internal/streams/readable:561:12)\n\tat readableAddChunkPushByteMode (node:internal/streams/readable:512:3)\n\tat Readable.push (node:internal/streams/readable:392:5)\n\tat Pipe.onStreamRead (node:internal/stream_base_commons:191:23)","category":"OpenOCDManager stderr","level":"error","timestamp":"2025-05-25T12:16:54.913Z"}
{"message":"OpenOCD server failed to start because of Error: Check JTAG interface, timings, target power, etc.\n❌ on failed: all zeroes\r\nError: Check JTAG interface, timings, target power, etc.\r\nError: Trying to use configured scan chain anyway...\r\nError: esp32s3.tap0: IR capture error; saw 0x00 not 0x01\r\nWarn : Bypassing JTAG setup events due to errors\r\n","stack":"Error: OpenOCD server failed to start because of Error: Check JTAG interface, timings, target power, etc.\n\tat Socket.<anonymous> (\\dist\\extension.js:2:1576120)\n\tat Socket.emit (node:events:524:28)\n\tat addChunk (node:internal/streams/readable:561:12)\n\tat readableAddChunkPushByteMode (node:internal/streams/readable:512:3)\n\tat Readable.push (node:internal/streams/readable:392:5)\n\tat Pipe.onStreamRead (node:internal/stream_base_commons:191:23)","category":"OpenOCDManager stderr","level":"error","timestamp":"2025-05-25T12:16:54.916Z"}
{"message":"OpenOCD server failed to start because of Error: [esp32s3.cpu0] Examination failed\n❌ Warn : [esp32s3.cpu0] Unexpected OCD_ID = 00000000\r\nError: [esp32s3.cpu0] Examination failed\r\nWarn : target esp32s3.cpu0 examination failed\r\nWarn : [esp32s3.cpu1] Unexpected OCD_ID = 00000000\r\nError: [esp32s3.cpu1] Examination failed\r\nWarn : target esp32s3.cpu1 examination failed\r\nInfo : [esp32s3.cpu0] starting gdb server on 3333\r\n","stack":"Error: OpenOCD server failed to start because of Error: [esp32s3.cpu0] Examination failed\n\tat Socket.<anonymous> (\\dist\\extension.js:2:1576120)\n\tat Socket.emit (node:events:524:28)\n\tat addChunk (node:internal/streams/readable:561:12)\n\tat readableAddChunkPushByteMode (node:internal/streams/readable:512:3)\n\tat Readable.push (node:internal/streams/readable:392:5)\n\tat Pipe.onStreamRead (node:internal/stream_base_commons:191:23)","category":"OpenOCDManager stderr","level":"error","timestamp":"2025-05-25T12:16:54.918Z"}
{"message":"OpenOCD server failed to start because of Error: JTAG tap: esp32s3.tap1  expected 1 of 1: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)\n❌ xffffffff (mfg: 0x7ff (<invalid>), part: 0xffff, ver: 0xf)\r\nError: JTAG tap: esp32s3.tap1  expected 1 of 1: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)\r\nError: Trying to use configured sc","stack":"Error: OpenOCD server failed to start because of Error: JTAG tap: esp32s3.tap1  expected 1 of 1: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)\n\tat Socket.<anonymous> (\\dist\\extension.js:2:1576120)\n\tat Socket.emit (node:events:524:28)\n\tat addChunk (node:internal/streams/readable:561:12)\n\tat readableAddChunkPushByteMode (node:internal/streams/readable:512:3)\n\tat Readable.push (node:internal/streams/readable:392:5)\n\tat Pipe.onStreamRead (node:internal/stream_base_commons:191:23)","category":"OpenOCDManager stderr","level":"error","timestamp":"2025-05-25T12:16:55.855Z"}
{"message":"OpenOCD server failed to start because of Error: esp32s3.tap0: IR capture error; saw 0x00 not 0x01\n❌ an chain anyway...\r\nError: esp32s3.tap0: IR capture error; saw 0x00 not 0x01\r\nWarn : Bypassing JTAG setup events due to errors\r\nWarn : [esp32s3.cpu0] Unexpected OCD_ID = 00000000\r\n","stack":"Error: OpenOCD server failed to start because of Error: esp32s3.tap0: IR capture error; saw 0x00 not 0x01\n\tat Socket.<anonymous> (\\dist\\extension.js:2:1576120)\n\tat Socket.emit (node:events:524:28)\n\tat addChunk (node:internal/streams/readable:561:12)\n\tat readableAddChunkPushByteMode (node:internal/streams/readable:512:3)\n\tat Readable.push (node:internal/streams/readable:392:5)\n\tat Pipe.onStreamRead (node:internal/stream_base_commons:191:23)","category":"OpenOCDManager stderr","level":"error","timestamp":"2025-05-25T12:16:55.856Z"}
{"message":"OpenOCD server failed to start because of Error: Target not examined yet\n❌ Warn : [esp32s3.cpu1] Unexpected OCD_ID = 00000000\r\nInfo : [esp32s3.cpu0] requesting target halt and executing a soft reset\r\nError: Target not examined yet\r\n","stack":"Error: OpenOCD server failed to start because of Error [... cut ...]
Error: [esp32c3] Could not identify target type.

For assistance with OpenOCD errors, please refer to our Troubleshooting FAQ: https://github.com/espressif/openocd-esp32/wiki/Troubleshooting-FAQ
OpenOCD Exit with non-zero error code 1
[Stopped] : OpenOCD Server
[/OpenOCD]
[Flash]
Can't perform JTag flash, because OpenOCD server is not running!

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

NoPlayBack, 2025-05-27:
Thanks for trying to help... unfortunately still no success here.

Ok, have now started all over from an Windows-11 Image in which i did never touch VSCode or USB-drivers.

My 2 variants for testing are
1. direct connection JTAG on ESP32C3-DevKitM-1 v1.6
- GND / 5V connected to USB GND / 5V
- USB-D+ connected to 19 / D- connected to 18
2. Lilygo T7-S3 connected on ESP-PROG
- GND / 5V connected to GND / 5V
- TMS / TCK / TDO / TDI connected, they have same naming on both boards.

Which one is better to test? I will proceed in this description with the variant 1.

Installed Microsoft Visual Studio Code 1.100.2
Installed  ESP-IDF v1.10.0 as extension in VSCode per description in https://github.com/espressif/vscode-esp-idf-extension/blob/master/README.md
Selected Github as downloadserver and v5.4.1 as ESP-IDF version
Did install FTDI-Driver CDM2123620_Setup.exe
Did connect ESP32C3-DevKitM-1 with the JTAG_lines wired to the USB to the windows machine.
Did select COM9 which was named to "ESP32-C3 (QFN32)(revision v.03)
Did select target as ESP32-C3 chip (via builtin USB-JTAG) ESP32-C3 debugging via builtin USB-JTAG (board/esp32c3-builtin.cfg)
configure your ESP-IDF: Did not change anything.

Result from first flash-trial was as expected again with error:

**************************************
debug_level: 2

Info : only one transport option; autoselecting 'jtag'

Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000

Info : Listening on port 6666 for tcl connections

Info : Listening on port 4444 for telnet connections

? Error: libusb_open() failed with LIBUSB_ERROR_NOT_FOUND
Error: esp_usb_jtag: could not find or open device!
C:\Users\Eddie\.espressif\tools\openocd-esp32\v0.12.0-esp32-20241016\openocd-esp32\share\o
? penocd\scripts/target/esp_common.cfg:9: Error: 
at file "C:\Users\Eddie\.espressif\tools\openocd-esp32\v0.12.0-esp32-20241016\openocd-esp32\share\openocd\scripts/target/esp_common.cfg", line 9
Error: [esp32c3] Unsupported DTM version: -1
Error: [
esp32c3] Could not identify target type.

For assistance with OpenOCD errors, please refer to our Troubleshooting FAQ: https://github.com/espressif/openocd-esp32/wiki/Troubleshooting-FAQ
OpenOCD Exit with non-zero error code 1
[Stopped] : OpenOCD Server
[/OpenOCD]
[Flash]
Can't perform JTag flash, because OpenOCD server is not running!

**************************************

Did try to switch to ESP-PROG-v1.0.0.0: Reaction from Windows was: Best drivers are already in place.

To answer your questions:
the full doctor command output did unfortunately not fit into the window where i should paste it, it gave an error "not longer than"... so i did select only the last part.

selected openOCD board configuration: ESP32-C3 chip (via builtin USB-JTAG) ESP32-C3 debugging via builtin USB-JTAG (board/esp32c3-builtin.cfg)

https://github.com/espressif/openocd-esp32/wiki/Troubleshooting-FAQ:
USB Adapter Is Not Connected would be the correct choice....
**How do i check with Device Dispatcher ???**
Did try IDF Tools installer, selected Universal Online Installer 2.3.5 windows 10,11. That one did install a lot of stuff into C:\Espressif\frameworks\esp-idf-v5.4.1

**Check DIS_USB_SERIAL_JTAG and DIS_USB_JTAG efuses. How do i do that ?**

https://docs.espressif.com/projects/esp-iot-solution/en/latest/hw-reference/ESP-Prog_guide.html
[... 212 lines cut ...]
t.c:1597 handle_target_init_command(): Initializing targets...
Debug: 111 35 riscv.c:449 riscv_init_target(): [esp32c3] riscv_init_target()
Debug: 112 36 semihosting_common.c:105 semihostin
g_common_init():  

? Error: 113 56 libusb_helper.c:278 jtag_libusb_open(): libusb_open() failed with LIBUSB_ERROR_N
? OT_FOUND
Error: 114 56 esp_usb_jtag.c:636 esp_usb_jtag_init(): esp_usb_jtag: could not find or open device!
Debug: 115 57 command.c:528 exec_command(): Command 'init' failed with error code -4
User : 116 57 command.c:600 command_run_line(): C:\Users\Eddie\.espressif\tools\openocd-esp32\v0.12.0-esp32-20241016\openocd-esp32\share\openocd\scripts/target/esp_common.cfg:9: Error: 
at file "C:\Users\Eddie\.espressif\tools\openocd-esp32\v0.12.0-esp32-20241016\openocd-esp32\share\openocd\scripts/target/esp_common.cfg", line 9
Debug: 117 57 breakpoints.c:328 breakpoint_remove_all_internal(): [esp32c3] Delete all breakpoints
Debug: 118 57 riscv.c:534 riscv_deinit_target(): [esp32c3] riscv_deinit_target()
Error: 119 57 riscv.c:429 get_target_type(): [esp32c3] Unsupported DTM version: -1
Error: 120 57 riscv.c:539 riscv_deinit_target(): [esp32c3] Could not identify target type.
Debug: 121 57 target.c:2139 target_free_all_working_areas_restore(): freeing all working areas

For assistance with OpenOCD errors, please refer to our Troubleshooting FAQ: https://github.com/espressif/openocd-esp32/wiki/Troubleshooting-FAQ
OpenOCD Exit with non-zero error code 1
[Stopped] : OpenOCD Server
[/OpenOCD]
[Flash]
Can't perform JTag flash, because OpenOCD server is not running!

**********************************************

NoPlayBack, 2025-05-27:
maybe one more comment....
I now did also use Zadeg to switch drivers.
Before doing that, i see 2 COM devices with FTDI drivers listed as driver.
After doing that, then one of the COM devices disappear and does not show up any more.
On the other hand... i did already contact Zadeg via Github and they confirmed that the exchange of the driver did work well (as per the log i did provide from that tool).....

NoPlayBack, 2025-05-30:
News:
Found out that the HW is not the issue.
Got it running on a test version of VisualGDB within VisualStudio2022.
The difference there is...
at installation and testing the connection, there was a message that USBDriverTool has to update the driver. Once that was done, there was no COM-port visible any more, instead there is a USB-Devide "USB JTAG/serial debug unit" visible. When selecting the device in VisualGDB, it does not offer a COM-port but it mentions directly the ESP32-device.
See screenshot attached.
So the HW works, and obviously the issue is the USB-driver.
Does this info help you, to help me, to get things running? VisualGDB is not the path i want to go...

[image]

brianignacio5, 2025-05-30:
Also consider that ESP32-C3 needs an special cable because the USB adapter CP2102N is an external serial converter not a USB_SERIAL_JTAG 

Detailed information here:

https://github.com/espressif/openocd-esp32/issues/343#issuecomment-2424735344

NoPlayBack, 2025-05-31:
did again reset my pc to original W11 installation without any VSCode or USB-driver stuff.
Did install Visual Studio Code, did install platformio, did try to debug, no luck.
Then, without changing driver with Zadig, did use 
Invoke-WebRequest 'https://dl.espressif.com/dl/idf-env/idf-env.exe' -OutFile .\idf-env.exe; .\idf-env.exe driver install --espressif
in order to clean up driver.
This is the output from the powershell:

`Invoke-WebRequest 'https://dl.espressif.com/dl/idf-env/idf-env.exe' -OutFile .\idf-env.exe; .\idf-env.exe driver install --espressif
Downloading https://dl.espressif.com/dl/idf-driver/idf-driver-esp32-usb-jtag-2021-07-15.zip to idf-driver-esp32-usb-jtag-2021-07-15.zip
* extracted: "C:\Users\Eddie/.espressif/tools/idf-driver/idf-driver-esp32-usb-jtag-2021-07-15\amd64/install-filter.exe" (44544 bytes)
* extracted: "C:\Users\Eddie/.espressif/tools/idf-driver/idf-driver-esp32-usb-jtag-2021-07-15\amd64/libusb0.dll" (76384 bytes)
* extracted: "C:\Users\Eddie/.espressif/tools/idf-driver/idf-driver-esp32-usb-jtag-2021-07-15\amd64/libusb0.sys" (52832 bytes)
* extracted: "C:\Users\Eddie/.espressif/tools/idf-driver/idf-driver-esp32-usb-jtag-2021-07-15\amd64/libusb0_x86.dll" (46592 bytes)
* extracted: "C:\Users\Eddie/.espressif/tools/idf-driver/idf-driver-esp32-usb-jtag-2021-07-15\amd64/libusbK.dll" (99128 bytes)
* extracted: "C:\Users\Eddie/.espressif/tools/idf-driver/idf-driver-esp32-usb-jtag-2021-07-15\amd64/libusbK.sys" (47928 bytes)
* extracted: "C:\Users\Eddie/.espressif/tools/idf-driver/idf-driver-esp32-usb-jtag-2021-07-15\amd64/libusbK_x86.dll" (84280 bytes)
* extracted: "C:\Users\Eddie/.espressif/tools/idf-driver/idf-driver-esp32-usb-jtag-2021-07-15\amd64/WdfCoInstaller01011.dll" (1795952 bytes)
* extracted: "C:\Users\Eddie/.espressif/tools/idf-driver/idf-driver-esp32-usb-jtag-2021-07-15\amd64/winusbcoinstaller2.dll" (1002728 bytes)
* extracted: "C:\Users\Eddie/.espressif/tools/idf-driver/idf-driver-esp32-usb-jtag-2021-07-15\USB_JTAG_debug_unit.cat" (9738 bytes)
* extracted: "C:\Users\Eddie/.espressif/tools/idf-driver/idf-driver-esp32-usb-jtag-2021-07-15\USB_JTAG_debug_unit.inf" (4642 bytes)
* extracted: "C:\Users\Eddie/.espressif/tools/idf-driver/idf-driver-esp32-usb-jtag-2021-07-15\x86/install-filter.exe" (46592 bytes)
* extracted: "C:\Users\Eddie/.espressif/tools/idf-driver/idf-driver-esp32-usb-jtag-2021-07-15\x86/libusb0.dll" (46592 bytes)
* extracted: "C:\Users\Eddie/.espressif/tools/idf-driver/idf-driver-esp32-usb-jtag-2021-07-15\x86/libusb0.sys" (42592 bytes)
* extracted: "C:\Users\Eddie/.espressif/tools/idf-driver/idf-driver-esp32-usb-jtag-2021-07-15\x86/libusb0_x86.dll" (67680 bytes)
* extracted: "C:\Users\Eddie/.espressif/tools/idf-driver/idf-driver-esp32-usb-jtag-2021-07-15\x86/libusbK.dll" (84280 bytes)
* extracted: "C:\Users\Eddie/.espressif/tools/idf-driver/idf-driver-esp32-usb-jtag-2021-07-15\x86/libusbK.sys" (42552 bytes)
* extracted: "C:\Users\Eddie/.espressif/tools/idf-driver/idf-driver-esp32-usb-jtag-2021-07-15\x86/WdfCoInstaller01011.dll" (1629040 bytes)
* extracted: "C:\Users\Eddie/.espressif/tools/idf-driver/idf-driver-esp32-usb-jtag-2021-07-15\x86/winusbcoinstaller2.dll" (851176 bytes)
Requesting elevation of privileges for: C:\Users\Eddie\idf-env.exe driver install --espressif --no-download
Exit code: 0x2a`

Still no success.
What does 0x2a mean ?

erhankur, 2025-06-06:
For the ESP-PROG, did you burn any efuse to use jtag pins instead of builtin usb jtag? 

https://docs.espressif.com/projects/esp-idf/en/stable/esp32s3/api-guides/jtag-debugging/configure-other-jtag.html#configure-efuses

NoPlayBack, 2025-06-07:
No, did not.
The Lilygo T7 S3 with an ESP32-S3-WROOM-1 on it has no GPIO19/20 available... instead it shows the 4 Pins named TDI, TDO, TCK, TMS. My asumption was that these are then permanently wired correctly ? Is this a wrong asumption?
Where are these efuses... within the ESP-chip or on the module?
Another question comes up when burning either DIS_USB_JTAG or STRAP_JTAG_SEL .... when first burning STRAP_JTAG_SEL ... would burning of DIS_USB_JTAG then "overwrite" the STRAP_JTAG_SEL setting ?

erhankur, 2025-06-07:
I’m not sure whether they were burnt during production. You can check `DIS_USB_JTAG` and  `STRAP_JTAG_SEL`  values first. 
```
espefuse.py -p /dev/tty.usbserial-210 summary
```

I guess both of them are false. If you burn the `STRAP_JTAG_SEL` you will need to connect GPIO3 to the ground. 
```
espefuse.py -p /dev/tty.usbserial-210 burn_efuse STRAP_JTAG_SEL 
```

Alternatively, you can burn the `DIS_USB_JTAG` fuse if you believe the USB-Serial-JTAG interface won’t be needed later. No additional pin is required with this option.
```
espefuse.py -p /dev/tty.usbserial-210 burn_efuse DIS_USB_JTAG 
```

NoPlayBack, 2025-06-08:
Need do add one comment to my statement from June 6.... did a complete fresh installation and did find out that one need first to use the command in a PowerShell:
`Invoke-WebRequest 'https://dl.espressif.com/dl/idf-env/idf-env.exe' -OutFile .\idf-env.exe; .\idf-env.exe driver install --espressif`
which is described here:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-guides/jtag-debugging/configure-builtin-jtag.html

...otherwise you do not have the option to change the driver to the "USB JTAG debug unit"...

NoPlayBack, 2025-06-16:
Update:
Tested JOY-IT SBC-NodeMCU-ESP32.
Did connect GPIO12..15 with ESP-PROG JTAG connections & a GND connection. ESP-PROG voltages switched to 3.3V.
Both, the NodeMCU and the ESP-PROG then need USB-connection to power up.
For this i did need ZADIG to get the drivers correct. "USB JTAG debug unit" was not available to select in the UsbDriverTool.exe, so i started with ZADIG in a first run.

After that it worked immediately.

brianignacio5, 2025-06-17:
Closing this issue as resolved. Comment here if you need more assistance.

closed 2025-06-17 by brianignacio5