espressif/vscode-esp-idf-extension #1591: [Feature Request]: When performing a Full Clean I would like it to also remove the 'dependencies.lock' file

AtkinEngineering, 2025-06-13:
Is it possible to make the Full Clean command, also remove the 'dependencies.lock' file?

brianignacio5, 2025-07-18:
Hi @AtkinEngineering sorry for late reply.

I've made PR #1613 which allows you to define files to be deleted on full clean command.

You can test these changes by installing this VSIX by click menu **View** -> **Command Palette...**, type **Install from VSIX** and then select downloaded **esp-idf-extension.vsix** file to install the extension.
* [esp-idf-extension.vsix.zip](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/3511626623.zip)

AtkinEngineering, 2025-07-20:
Hi @brianignacio5 thank you for this. Unfortunately i can not get it to work. using `"idf.extraCleanPaths": ["managed_components","dependencies.lock"],` has an error of `Incorrect type. Expected "string[]".`

AtkinEngineering, 2025-07-30:
Hi @brianignacio5 thank you for this. The error has gone and it does delete the managed components folder, however it doesn't seem to delete the dependencies.lock file.

brianignacio5, 2025-07-31:
Hi @AtkinEngineering Apologies  I found the issue on my part I was just checking for existing folders and files were being left out.

I've updated the PR and this is the [latest vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/3646576252.zip) with fixes.

closed 2025-07-31 by brianignacio5

AtkinEngineering, 2025-08-01:
Hi @brianignacio5 Thank you for this. This works as you described.