espressif/vscode-esp-idf-extension #1602: [Feature Request]: Auto Port Selection

14790897, 2025-07-01:
**Describe the solution you'd like**
I'd like to request an **auto port selection feature**, which could either automatically detect the correct device based on known criteria

brianignacio5, 2025-07-01:
What is the known criteria? 

esptool.py can do auto detect but it would try multiple serial ports until it finds one that matches the current IDF_TARGET. What about when user has multiple devices of same IDF_TARGET ?

We could add this feature, but the auto-detect would make it slower and if multiple devices of same IDF_TARGET, the wrong one could be flashed.

14790897, 2025-07-01:
It would be best to add an option allowing users to choose whether to enable it or not.

closed 2025-08-20 by github-actions[bot]