espressif/vscode-esp-idf-extension #1610: Cannot using debugging feature on VSCode extention for ESP32-S3-MINI-1

eren-rudy, 2025-07-01:
### OS

MacOS

### Operating System version

macOS 13.7.4

### Visual Studio Code version

1.100.3

### ESP-IDF version

1.10.1

### Python version

3.10.9

### Doctor command output

[doctor.txt](https://github.com/user-attachments/files/21007455/doctor.txt)

### Extension

[esp_idf_vsc_ext.log](https://github.com/user-attachments/files/21007457/esp_idf_vsc_ext.log)

### Description

Trying to debug an ESP32-S3-MINI-1 on a custom board via it's built-in USB-JTAG interface. Flashing and monitoring works as expected but attempting to debug does not. OpenOCD reports that both cores have been halted, then nothing else happens.

### Debug Message

Verbose bugs are provided below, but a summary of what I saw without verbose logging:
- OpenOCD reported successful connection of device, followed by both cores being halted. I saw a few different reasons given on different attempts (WDT reset, RTC reset, software reset, USB/UART reset...), but could not repeat this
- GDB logs report "missing symbol called"

[openocd.txt](https://github.com/user-attachments/files/21007461/openocd.txt)

[gdb.txt](https://github.com/user-attachments/files/21007462/gdb.txt)

### Other Steps to Reproduce

Also tried -- without success -- running OpenOCD on the command line, providing interface and target files explicitly:
`openocd -f interface/esp_usb_jtag.cfg -f target/esp32s3.cfg `

Checked that the JTAG/Serial debug unit is detected with `system_profiler SPUSBDataType` on terminal:
``
USB 3.1 Bus:

      Host Controller Driver: AppleT6000USBXHCI

        USB JTAG/serial debug unit:

          Product ID: 0x1001
          Vendor ID: 0x303a
          Version: 1.01
          Serial Number: CC:8D:A2:20:24:68
          Speed: Up to 12 Mb/s
          Manufacturer: Espressif
          Location ID: 0x01100000 / 2
          Current Available (mA): 500
          Current Required (mA): 500
          Extra Operating Current (mA): 0
``

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2025-07-04:
This seems like an issue with the openOCD configuration.

Please review the [OpenOCD Troubleshoot FAQ](https://github.com/espressif/openocd-esp32/wiki/Troubleshooting-FAQ#esp32-c3-and-esp32-s3-builtin-usb_jtag-gets-stuck-after-hw-reset)

You can also open an issue there, If there something on the vscode extension side I'm happy to assist.

brianignacio5, 2025-07-18:
Closing this issue for lack of response from user.

Comment here if you need more assistance.