espressif/vscode-esp-idf-extension #1620: JTAG flash TCL + WINDOWS

Maddocrus, 2025-07-17:
### OS

Windows

### Operating System version

Windows 10

### Visual Studio Code version

1.102

### ESP-IDF version

5.3.1

### Python version

3.8.7

### Doctor command output

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.19045 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:\Program Files (x86)\Common Files\Oracle\Java\java8path;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\MadProjects\Espressif\.esp-idf\tools;C:\msys64\mingw64\bin;C:\Program Files (x86)\Microchip\MPLABX\v5.30\gnuBins\GnuWin32\bin\;C:\MCC18\mpasm;C:\MCC18\bin;C:\Program Files (x86)\Windows Kits\10\Windows Performance Toolkit\;C:\Program Files (x86)\STMicroelectronics\STM32 ST-LINK Utility\ST-LINK Utility;C:\Program Files\Git\cmd;C:\Program Files\dotnet\;C:\Program Files (x86)\Microchip\xc32\v1.32\bin;C:\Program Files\Microchip\xc32\v1.32\bin;C:\Program Files\LLVM\bin;C:\Windows\System32;C:\Program Files\PowerShell\7\;C:<HOMEPATH>\.dotnet\tools;C:<HOMEPATH>\AppData\Local\Programs\Microsoft VS Code\bin;C:\MadProjects\Espressif\.esp-idf\tools\openocd-esp32\v0.12.0-esp32-20240318\openocd-esp32\bin;C:\Program Files\CMake\bin;C:\Program Files (x86)\Arm GNU Toolchain arm-none-eabi\14.2 rel1\bin\;c:\openocd\bin\;C:\Program Files\Docker\Docker\resources\bin;C:\Program Files (x86)\dotnet\;C:\Program Files\Go\bin;C:\Program Files\PuTTY\;C:<HOMEPATH>\AppData\Local\Programs\Python\Launcher\;C:\Program Files (x86)\Arm GNU Toolchain arm-none-eabi\14.2 rel1\bin;C:\msys64\mingw64\bin;C:\GnuWin32\bin;c:\Program Files (x86)\Microchip\MPLABX\v5.30\gnuBins\GnuWin32\bin\;C:\MCC18\mpasm;C:\MCC18\bin;C:<HOMEPATH>\.dotnet\tools;C:<HOMEPATH>\AppData\Local\Programs\Microsoft VS Code\bin;C:\Windows\System32;C:<HOMEPATH>\go\bin 
System environment variable PYTHON 
 undefined 
Visual Studio Code Remote name undefined 
Visual Studio Code version 1.102.0 
Visual Studio Code language en 
Visual Studio Code shell C:\Program Files\PowerShell\7\pwsh.exe 
ESP-IDF Extension version 1.10.5 
Workspace folder c:\MadProjects\Espressif\power_save 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) c:\MadProjects\Espressif\.esp-idf\frameworks\esp-idf-v5.3.1
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
ESP-HomeKit-SDK Path (idf.espHomeKitSdkPath) ${env:HOMEKIT_PATH}
Custom extra paths c:\MadProjects\Espressif\.esp-idf\tools\tools\xtensa-esp-elf-gdb\14.2_20240403\xtensa-esp-elf-gdb\bin;c:\MadProjects\Espressif\.esp-idf\tools\tools\riscv32-esp-elf-gdb\14.2_20240403\riscv32-esp-elf-gdb\bin;c:\MadProjects\Espressif\.esp-idf\tools\tools\xtensa-esp-elf\esp-13.2.0_20240530\xtensa-esp-elf\bin;c:\MadProjects\Espressif\.esp-idf\tools\tools\riscv32-esp-elf\esp-13.2.0_20240530\riscv32-esp-elf\bin;c:\MadProjects\Espressif\.esp-idf\tools\tools\esp32ulp-elf\2.38_20240113\esp32ulp-elf\bin;c:\MadProjects\Espressif\.esp-idf\tools\tools\cmake\3.24.0\bin;c:\MadProjects\Espressif\.esp-idf\tools\tools\openocd-esp32\v0.12.0-esp32-20240318\openocd-esp32\bin;c:\MadProjects\Espressif\.esp-idf\tools\tools\ninja\1.11.1;c:\MadProjects\Espressif\.esp-idf\tools\tools\idf-exe\1.0.3;c:\MadProjects\Espressif\.esp-idf\tools\tools\ccache\4.8\ccache-4.8-windows-x86_64;c:\MadProjects\Espressif\.esp-idf\tools\tools\dfu-util\0.11\dfu-util-0.11-win64;c:\MadProjects\Espressif\.esp-idf\tools\tools\esp-rom-elfs\20240305
ESP-IDF extra vars
    OPENOCD_SCRIPTS: c:\MadProjects\Espressif\.esp-idf\tools\tools\openocd-esp32\v0.12.0-esp32-20240318\openocd-esp32\share\openocd\scripts
    IDF_CCACHE_ENABLE: 1
    ESP_ROM_ELF_DIR: c:\MadProjects\Espressif\.esp-idf\tools\tools\esp-rom-elfs\20240305\
    IDF_PYTHON_ENV_PATH: c:\MadProjects\Espressif\.esp-idf\tools\python_env\idf5.3_py3.11_env
    ESP_IDF_VERSION: 5.3
    PATH: c:\MadProjects\Espressif\.esp-idf\tools\tools\xtensa-esp-elf-gdb\14.2_20240403\xtensa-esp-elf-gdb\bin;c:\MadProjects\Espressif\.esp-idf\tools\tools\riscv32-esp-elf-gdb\14.2_20240403\riscv32-esp-elf-gdb\bin;c:\MadProjects\Espressif\.esp-idf\tools\tools\xtensa-esp-elf\esp-13.2.0_20240530\xtensa-esp-elf\bin;c:\MadProjects\Espressif\.esp-idf\tools\tools\riscv32-esp-elf\esp-13.2.0_20240530\riscv32-esp-elf\bin;c:\MadProjects\Espressif\.esp-idf\tools\tools\esp32ulp-elf\2.38_20240113\esp32ulp-elf\bin;c:\MadProjects\Espressif\.esp-idf\tools\tools\cmake\3.24.0\bin;c:\MadProjects\Espressif\.esp-idf\tools\tools\openocd-esp32\v0.12.0-esp32-20240318\openocd-esp32\bin;c:\MadProjects\Espressif\.esp-idf\tools\tools\ninja\1.11.1\;c:\MadProjects\Espressif\.esp-idf\tools\tools\idf-exe\1.0.3\;c:\MadProjects\Espressif\.esp-idf\tools\tools\ccache\4.8\ccache-4.8-windows-x86_64;c:\MadProjects\Espressif\.esp-idf\tools\tools\dfu-util\0.11\dfu-util-0.11-win64;c:\MadProjects\Espressif\.esp-idf\tools\python_env\idf5.3_py3.11_env\Scripts;c:\MadProjects\Espressif\.esp-idf\frameworks\esp-idf-v5.3.1\tools;%PATH%
    IDF_DEACTIVATE_FILE_PATH: C:<HOMEPATH>\AppData\Local\Temp\tmpefdsk9_vidf_26388
User extra vars (idf.customExtraVars)
System python Path (idf.pythonInstallPath) c:\MadProjects\Espressif\.esp-idf\tools\tools\idf-python\3.11.2\python.exe
Virtual environment Python path (computed) c:\MadProjects\Espressif\.esp-idf\tools\python_env\idf5.3_py3.11_env\Scripts\python.exe
Serial port (idf.port) COM9
OpenOCD Configs (idf.openOcdConfigs) board/esp32c6-builtin.cfg
OpenOCD log level (idf.openOcdDebugLevel) 2
OpenOCD launch arguments (idf.openOcdLaunchArgs) 
ESP-IDF Tools Path (idf.toolsPath) c:\MadProjects\Espressif\.esp-idf\tools
Git Path (idf.gitPath) c:\MadProjects\Espressif\.esp-idf\tools\tools\idf-git\2.39.2\cmd\git.exe
Notification Mode (idf.notificationMode) Output
Flash type (idf.flashType) UART
Flash partition to use (idf.flashPartitionToUse) all

### Description

If I flash via UART and debug via JTAG, then everything works, but I need to flash and debug via JTAG.
When i'm trying to flash through JTAG then i get an error: 
```
❌ 20240318\openocd-esp32\share\openocd\scripts/target/esp_common.cfg:519: Error: c:LoraWanlePowerlinkTestproject/build/flasher_args.json: Invalid argument
in procedure 'program_esp_bins' 
```
c:LoraWanlePowerlinkTestproject - TCL script does not correctly handle the path and treats "\\" as a control character.
If i corrected the esp_common.cfg:519: file to a specific path in "proc program_esp_bins {build_dir filename args}"
```
	set build_dir "C:\\LoraWan\\blePower\\blinkTest\\project\\build"
	set fp [open [file join $build_dir $filename] r]
```
Then this section of the code runs fine, but there's an error again.
```
Info : Auto-detected flash bank 'esp32c6
.flash' size 8192 KB
Info : Using flash bank 'esp32c6.flash' size 8192 KB

❌ Error: couldn't open C:LoraWanlePowerlinkTes
tprojectuild/bootloader/bootloader.bin
** Flashing Failed **
```
But If i corrected the esp_common.cfg:519: file to a specific path in "proc program_esp_bins {build_dir filename args}"
```
	set build_dir "C:/LoraWan/blePower/blinkTest/project/build"
	set fp [open [file join $build_dir $filename] r]
```
Flashed Successfully (JTAG)!

I expect that the path will be processed correctly.
c:/LoraWan/blePower/blinkTest/project/build/flasher_args.json:
and it will be possible to flash via JTAG on WINDOWS + VSCode. The esp-idf Eclipse firmware is working. I can't track what the VSCode extension is sending to TCL script.

### Debug Message

up

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2025-07-17:
there is an extension setting `openocd.jtag.command.force_unix_path_separator` that forces a replacement from \\ to / for build path.

You can test using `"openocd.jtag.command.force_unix_path_separator": false` in your project .vscode/settings.json to see if it works.

Maddocrus, 2025-07-17:
Thanks it helped ->  "openocd.jtag.command.force_unix_path_separator": true,