espressif/vscode-esp-idf-extension #1628: Debugging Support on ESP32-P4 Functional EV Board (v1.5.2)

sachinraghugowda, 2025-07-24:
### OS

Windows

### Operating System version

windows 11

### ESP-IDF version

5.5.0

### Python version

3.11.0

### Doctor command output

[doctor_command_output.zip](https://github.com/user-attachments/files/21414761/doctor_command_output.zip)

### Extension

[esp_idf_vsc_ext_log.zip](https://github.com/user-attachments/files/21414768/esp_idf_vsc_ext_log.zip)

### Description

I am using the new version (1.5.2) ESP32 P4 FUnctional EV kit, and and attempting to enable debugging via the inbuilt USB Serial JTAG interface(port) using Visual Studio Code and the ESP-IDF environment.

I have followed the steps outlined in your documentation for JTAG debugging:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32p4/api-guides/jtag-debugging/index.html#jtag-debugging-configure-and-connect-jtag-interface

I was able to run OpenOCD successfully and reached the expected output as mentioned. Please refer the attached image from my terminal output:

[image]

Next step, I proceeded with configuring debugging in Visual Studio Code, following the guide below:
https://docs.espressif.com/projects/vscode-esp-idf-extension/en/latest/debugproject.html
The required drivers were installed automatically. However, when I attempt to start the debugger (after selecting the appropriate board/esp32p4-builtin.cfg), I encounter persistent errors in the output window as below :

[image]

 Could you help me identify any potential misconfigurations or limitations that may be causing this issue?

### Debug Message

Debug adapter -> Extension: DEBUG_ADAPTER_STARTED
Debug adapter -> Extension: DEBUG_ADAPTER_READY2CONNECT

### Other Steps to Reproduce

I tried chaning the launchs to esp-IDF launch and Eclipse CDT GDB Launch. each resukted in these error

In have also attached my settings.json file here for your reference

[settings.json](https://github.com/user-attachments/files/21414774/settings.json)

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

sachinraghugowda, 2025-07-30:
Hi @sobuch I tried with the new release version [openocd-esp32-win64-0.12.0-esp32-20250707], still kind of same error. 
after that when i run debug i get below error in the output. It is in the attachement:

[OpenOcd output file_new version.txt](https://github.com/user-attachments/files/21504898/OpenOcd.output.file_new.version.txt)

below is my terminal output: 

[image]

It still says memory protection not supported yet. Do you mean to use any other version of openOCD?

I am also attaching the laund.json and settings.json here for your conveience. 

[launch.json](https://github.com/user-attachments/files/21504928/launch.json)
[settings.json](https://github.com/user-attachments/files/21504929/settings.json)

Please let me know, if I am missing something still.

sobuch, 2025-08-02:
Hi @sachinraghugowda there is probably version mismatch of OpenOCD binaries and config files, see in the output file:
```
Open On-Chip Debugger v0.12.0-esp32-20241016 (2024-10-16-14:17)
...
Debug: 23 10 configuration.c:88 find_file(): found C:\Users\DE10K21008\.espressif\tools\openocd-esp32\v0.12.0-esp32-20250707/openocd-esp32/share/openocd/scripts/board/esp32p4-builtin.cfg
```
The warning "Memory protection support is not implemented yet" also means outdated config files are used.

I am not actually sure how to select OpenOCD binary  in the project settings, so could you try like this:
- make sure OpenOCD server is stopped in the vscode
- run the new release version in terminal `openocd -c "set ESP_ONLYCPU 0x1"  -f board/esp32p4-builtin.cfg -d3 -l openocd_log.txt`
- with the OpenOCD running, start debug session in vscode
- in case that doesnt work, please share log file openocd_log.txt

sobuch, 2025-08-13:
Hi @sgowda95 , thanks for your efforts! Unfortunately I still dont see anything in the logs. But there are some more things we can try:

- reset the target before GDB attempts to connect:

`openocd -c "set ESP_ONLYCPU 0x1"  -f board/esp32p4-builtin.cfg -d3 -l openocd_log.txt -c "init;reset halt"`
 
- enable logging from the flasher stub:

`openocd -c "set ESP_ONLYCPU 0x1"  -f board/esp32p4-builtin.cfg -d3 -l openocd_log.txt -c "esp stub_log on"`
 
- disable flash support, note that this will impose a limit of max 3 hw breakpoints:

`openocd -c "set ESP_ONLYCPU 0x1; set ESP_FLASH_SIZE 0"  -f board/esp32p4-builtin.cfg -d3 -l openocd_log.txt`

Alternatively, would you be able to share an example project with us, so we can try reproduce the issue?
Also, have you tried whether debugging works for you with some simple examples from esp-idf? If not please give it a try, ideally also with the `CONFIG_FREERTOS_UNICORE=y` setting.

sgowda95, 2025-08-15:
Hi @sobuch I tried all the above 3 cases, but still was not able to get the debug working. it was still more or less the same error which i have attached above

However I tried also debugging a simple hello_world example form the esp-idf examples and ended up facing the same issue, could not get the deubgger running for a hello_world example either. 

I am attching the hello_world example which i tried. However, I have replaced the launch.json and settings.json from this hello_world example to that of the configrations i am using in my original project. Also I did change the CONFIG_FREERTOS_UNICORE=y for  this example and also have a look at the openocd_log.txt in the proeject for the error log files.  I am attaching this example here. please have a look and let me know, if i am missing something. 

[hello_world.zip](https://github.com/user-attachments/files/21794474/hello_world.zip)

sobuch, 2025-09-01:
Not stale. I was not able to reproduce exactly the same thing, but I saw a similar issue in different scenarios.

We will need to fix that first, and then can check whether it helps here as well.

sobuch, 2025-09-30:
Hi @sgowda95  , we are testing a workaround for a similar issue with esp32p4, would you be able to check if this version also solves your issue?

This version also already supports debugging with an unavailable cpu1 target, so it should not be needed anymore to set the ESP_ONLYCPU variable. Please see prebuild binaries in the attached archive, and I also pushed the sources in https://github.com/sobuch/openocd-esp32/tree/esp32p4_dowload_mode_flasher_fix

[openocd-esp32.zip](https://github.com/user-attachments/files/22618136/openocd-esp32.zip)

brianignacio5, 2026-01-14:
Can we close this issue @sobuch @sgowda95 @sachinraghugowda ?